linalinn/kicad-render #1: Make work with KiCad 8

dBitech, 2024-03-26:


linalinn, 2024-03-26:
Hii,
I have seen you removed the spaces from the files but not from the 3d-images.yml which should fix it. However, I fixed the original problem that the action did not handle paths containing space correctly.
Let me know if there is still a problem.

linalinn, 2024-04-10:
Hii,
since there is no activity here I am closing this Issue the problem has been fixed.
The commit is [here](https://github.com/linalinn/kicad-render/commit/31bef9de1e05c40dc25da704ac698ab4a99c476f).

closed 2024-04-10 by linalinn

VA7DBI, 2024-04-10:
That Fix does not address all of the issues, as you can see the "Unknown file type" [error](https://github.com/VA7DBI/ideal-or-diode/actions/runs/8638160821/job/23681980082#:~:text=8-,rendering%20top,21%3A41%3A22%3A%20Error%3A%20Unknown%20file%20type,-13):    https://github.com/VA7DBI/ideal-or-diode/actions/runs/8638160821/job/23681980082