linalinn/kicad-render #3: Option to configure animation output resolution (WIDTH, HEIGHT)

JakubAndrysek, 2024-04-14:
Option to configure animation output resolution (WIDTH, HEIGHT)

Can you please add an option to configure the [animation output](https://github.com/linalinn/kicad-render/blob/main/kicad_animation.sh#L32-L33) resolution?
- WIDTH=300
- HEIGHT=300

Furthermore, I would like to configure [all options](https://github.com/linalinn/kicad-render/blob/main/kicad_animation.sh#L31-L38) from `.yaml` file.

Thank you for the great project 😍

linalinn, 2024-04-19:
Hi, 
to add the ability to change the WIDTH and HIGHT I already thought about so that would be doable like soonish.
One question I have is which benefit you see if the config would be loaded from a `.yaml` file instead of from the workflow.

JakubAndrysek, 2024-04-19:
Oh, sorry, I thought the workflow file ending with a `.yaml` extension.