linalinn/kicad-render #13: Error on workflow run

elpiel, 2025-01-16:
For some reason I get this error when running the workflow and can't figure out while
```
 Options: -c 'export.kibot.yaml' -d 'output' -b 'XXX-MVP.kicad_pcb' -e 'XXX-MVP.kicad_sch'
- 'Print schematic (PDF)' (print_sch) [pdf_sch_print]
WARNING:(W008) Unable to find KiCad configuration file (/github/home/.config/kicad/7.0/kicad_common.json) (kibot - config.py:200)
ERROR:Running ['kicad-cli', 'sch', 'export', 'pdf', '-o', '/github/workspace/output/XXX-MVP.pdf', '-t', '_builtin_default', '--no-background-color', '/github/workspace/XXX-MVP.kicad_sch'] returned 3 (eeschema_do.kiauto.file_util - file_util.py:390) (kibot - kiplot.py:130)
ERROR:/usr/bin/eeschema_do returned 19 (kibot.gs - gs.py:850)
```

 I can share my configu file as well.

elpiel, 2025-01-16:
Sorry wrong repo, wanted to suggest something else for kicad-render: custom Fonts.
Let me quickly fix that.

linalinn, 2025-01-18:
Hi,
yes it is not able to find it because it's inside of the container.
I'll gonna take a look at it as soon as possible sadly I can't provide an easy hack as a workaround.

elpiel, 2025-01-20:
Maybe some pre-installed fonts for starters?

linalinn, 2025-04-10:
Hii,
yes and that is what it is for (I am planning to make a proper release) but for now, I can point you to my test setup which includes installing MS fonts.
[script](https://github.com/linalinn/kicad-render/blob/b16f0d2ce791c1b70d436f55aea0e568c4e2d602/.github/tests/test%20pcb/kicad_render_pre_render.sh#L6) (Line: 6 to 10)
[workflow](https://github.com/linalinn/kicad-render/blob/b16f0d2ce791c1b70d436f55aea0e568c4e2d602/.github/workflows/image.yaml#L89) (Line: 89 to 95)

Oh, I just checked the readme and there is already some documentation.

TL;DR
The links and readme should help you to install MS fonts. Just make sure to use `linalinn/kicad-render@main`.

elpiel, 2025-04-10:
Already using `main`

closed 2025-05-07 by elpiel